brianinnes/vPiP #1: Servo height settings are ignored

MarkJB, 2016-03-06:
Running test2.py and modifying the config values for penUp and penDown doesn't appear to have any effect on where the servo goes.

I tried a range of values but it doesn't change the servo positions.

I set them both to 0 and its still the same.

I changed the values in the Arduino firmware from:

```const long PENUP_ANGLE = 170;
const long PENDOWN_ANGLE = 85;```

to

```const long PENUP_ANGLE = 125;
const long PENDOWN_ANGLE = 63;```

and the positions changed.

This was with Python 2.7.11 on Win7x64

MarkJB, 2016-03-06:
I don't know why there aren't line breaks in the above... Markdown never works properly for me.

MarkJB, 2016-03-06:
If you are wondering the python is seeing the config:

C:\Users\Mark\Documents\GitHub\pycupi\python>c:\Python\Python27-11\python.exe test2.py
Polargraph __init__
Config is being read from C:\Users\Mark/.polargraph/config.cfg

***** System configuration *****
configured = True
penSize = 0.5
machineWidth = 610
machineHeight = 1010
mmPerRev = 92.0
stepsPerRev = 200
stepMultiplier = 8
serialPort = COM10
timeSliceUS = 2048.0
baud = 57600
motorAccel = 0.2
motorMaxSpeed = 100.0
penUp = 0
penDown = 0
homeX = 305
homeY = 120
polarDraw = True
size = A3p
width = 420
height = 297
posX = 83
posY = 398
margin = 20
pixelsX = 5000
pixelsY = 3381
rotate = False
stepsSizeMM = 0.0575
stepsPerValue = 15.75
MaxSpeedMMs = 100.0
AccelerationMMs2 = 500.0
screenX = 5000
screenY = 3381
showImage = False
saveImage = False
Coord handler running

Step handler running
Connected to Polargraph

Polargraph __exit__

brianinnes, 2016-03-06:
Ah, I forgot about this - it is on the todo list.  The gocupi protocol currently does not provide any options to send config data from the controlling device to the Arduino.  I have a cunning plan to use the reset command to be able send additional config data whilst still being able to run against a true gocupi system (where the config for micro-steps, timing window and penup/pen down angles all need to be set in Arduino code, and where applicable also in the controller code).

brianinnes, 2017-09-10:
Implemented in v0.2 in commit 37af6e78de0516c9e5431b9141468f667e025eab

closed 2017-09-10 by brianinnes

MarkJB, 2017-09-10:
Liking the config! 

Very cool that you can define the hardware pins from the controller rather than the firmware.

Am I right in thinking the config is extensible just be virtue of adding items to the config stuct and then send the relevant data (in order)?

brianinnes, 2017-09-10:
Yes, so long as the structure on the Pi and the Arduino are kept in sync you can add additional attributes and they will be sent to the Arduino on startup - the entire structure is sent to the Arduino.

You need to be aware of the structure packing differences between 64 and 8 bit architectures - hence the pack compiler instruction.

MarkJB, 2017-09-11:
One thing just struck me, https://github.com/brianinnes/vPiP/issues/16 says v2 must work with existing hardware which would mean being able to use serial if I2C isn't available/connected.