brianinnes/vPiP #4: Drawing doesn't start

MarkJB, 2016-03-13:
Occasionally when starting a plot, the drawing doesn't start. 

I see the config output and the servo does its 'up - down - up' move and then it hangs.

I have to pkill the python processes to start again.

MarkJB, 2016-03-21:
I haven't seen this happen for a while now.

Maybe it was environmental.

If I don't see it happening after a few more drawings i'll close the issue.

MarkJB, 2016-03-22:
I take that back... seen this today on a vector.

brianinnes, 2016-03-23:
What were the circumstances when it didn't draw?  
Was the previous draw aborted? - trying to determine what may cause the freeze up

MarkJB, 2016-03-23:
This was after stopping a previous run with 'pkill python'.

MarkJB, 2016-05-09:
It would be good to get some logging into vPiP... I'm seeing this with a jpg now.

MarkJB, 2016-05-10:
Ok, that was my own fault. The image I was trying to plot had been converted to an indexed image at one point and then back to RGB and the assigned RGB value was so low vPiP figured it was a blank image. A test render confirmed it. There was nothing to draw...

Logging is still a good idea though!