brianinnes/vPiP #10: Units should be in mm

MarkJB, 2016-04-19:
This is an Enhancement request. Might be irrelevant in light of the pending work, but raising it anyway as it is worth some consideration.

It is difficult to work in pixels when using multiple machines with different page sizes.

It would make life easier if units were in mm.

brianinnes, 2016-04-19:
In the config just set the pixels value to the same as the paper width (in mm) then the measurements are in mm.

closed 2016-04-19 by brianinnes

MarkJB, 2016-04-19:
Does that not affect the 'resolution' of the pixels?

brianinnes, 2016-04-19:
depends.  If you always make the pixel size 1 mm square (pixel size in line with the paper size), then the number of pixels in each direction will change as paper size changes.  If you leave the pixel size the same across all the different paper size, then each paper size will have a scaled pixel size, but there will be the same number of pixels across the page.

Current implementation allows either way of working, but some of my test apps assume a 5000 pixel width.