brianinnes/vPiP #9: Auto centre images

MarkJB, 2016-04-19:
This is an Enhancement request. Might be irrelevant in light of the pending work, but raising it anyway as it is worth some consideration.

Software needs to be able to auto centre an image.

brianinnes, 2016-04-19:
Need to clarify the requirement as the current renderer implementation allows you to specify where to place the top left corner of the image and the required width to draw the image which scales the image accordingly.  Easy to calculate these values when you want to position an image at the centre of the page at the scale you want as you have the image size and can get the paper size from config.

MarkJB, 2016-04-19:
I would suggest if x & y are not given (or maybe set to -1) that x & y are calculated automatically given the supplied image dimension (i.e. width).

I guess x would be (p.width - image.width)/2 and y would be (p.height - image.height)/2

brianinnes, 2016-04-19:
yes.  There are a number of other 'helper' functions that could be part of the library, such as 'scale image to fit on page and centre 1 axis if required'

The constrainDrawingRectangle can also be classified as a helper function.

We should define required helper functions and work out how to get them to be applied to a renderer without making the API too complex.

We could come up with a system where we say 'within these **constraints** render an image using this **renderer** and this **pixel**', where the helper functions form a set of constraints.