brianinnes/vPiP #19: Create documentation on setting up Mosquitto message broker

brianinnes, 2017-04-27:


brianinnes, 2017-04-27:
Assumption is that this will initial run on a Raspberry Pi - is this correct?
Do we need a Docker container for running on laptops?
Do we need to support a cloud service?

MarkJB, 2017-04-27:
It would be useful to be able to run the software on a laptop/PC as well as the Raspberry Pi.

Assuming the rest of the software can be made to run on a PC as well (i.e. the serial comms module). Is the software tied to a specific message broker in any way? (i.e. can we not just swap out one message broker for another).

brianinnes, 2017-04-27:
Mosquitto can run on Windows, macOS and Linux.  Unfortunately I don't think there is a universal API that works across message brokers, so focus will be the defacto standard, which is MQTT.  This is what Mosquitto supports in addition to a number of [other servers](https://github.com/mqtt/mqtt.github.io/wiki/servers).  It should be possible to swap out Mosquitto for another server (so long as they don't impose any topic namespace restrictions)

For the stepper driver It is probably safe to assume that other Linux systems will be able to run the software, but would need to add additional build steps (Vagrant build machine is Debian, so all the build tools for 32/64 bit Linux builds are already installed).
Would need to look at comms handling in macOS and Windows to see if they can be supported - the primary goal will be the Raspberry Pi, other platforms will be secondary

brianinnes, 2017-04-27:
Mosquitto can run on Windows, macOS and Linux.  There is no universal message bus API, but MQTT is seen as a defacto standard, which is what Mosquitto supports, in addition to a number of [other servers](https://github.com/mqtt/mqtt.github.io/wiki/servers), so it should be possible to switch out Mosquitto for one of the other brokers (so long as there are no topic space restrictions which are incompatible to our implementation).   

For the stepperDrivers it should be possible to run the code on a Linux system.  The build system is based on Debian, so adding i386/x64 builds shouldn't be too difficult.  Will need to look at Windows / macOS comms systems to determine how difficult it will be to support these.   

Focus of the initial project will be getting everything working on Raspberry Pi, other platforms will be secondary